WPIRoboticsEngineering/ZenDriver #8: Add the line driver chip to Serial2 (pins 16 and 17) using Pin 19 as line enable for the LewanSoul motors

madhephaestus, 2020-10-27:
I tested this out with the baby yoda and it worked well. It would allow us to add arbitrarily more motors to the 1001 kits using the bus style connections. 

Add 3 sockets to the bus to distribute the current load.

acamilo, 2020-10-27:
Is there another set of pins we can use?
16 and 17 are going to the LIDAR connector.  It would be nice to have both lidar and LS motors.

madhephaestus, 2020-10-27:
same pins for the same reason, its the only availible hardware serial port

acamilo, 2020-10-27:
You can use the GPIO matrix to route the uart pins to free GPIO.

acamilo, 2020-10-27:
See chapter 5 of this document
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

acamilo, 2020-10-27:
You might be able to mux TX and RX as well without an external chip.

madhephaestus, 2020-10-27:
yeah i was looking into that, ill try it on Yoda

acamilo, 2020-10-28:
What power rail should I use? 5v or battery?

madhephaestus, 2020-10-29:
battery

acamilo, 2020-10-31:
Added 3 lewansoul connectors to rev 2. No buffer, directly connected to GPIO33

closed 2020-10-31 by acamilo