WPIRoboticsEngineering/ZenDriver #16: Swap encoder GPIO.

acamilo, 2020-10-31:
A and B are swapped on both encoders
encoder 1 and 2 need to be swapped.

This is the config that works: 
#define MOTOR_RIGHT_ENCB 26
#define MOTOR_RIGHT_ENCA 27
#define MOTOR_LEFT_ENCB 32
#define MOTOR_LEFT_ENCA 14

It needs to be this
#define MOTOR_LEFT_ENCA 26
#define MOTOR_LEFT_ENCB 27
#define MOTOR_RIGHT_ENCA 32
#define MOTOR_RIGHT_ENCB 14

acamilo, 2020-11-01:
in 0.9.3/RBE1001.h the left encoder pinout matches BOTH the shim and the zen
but the right side has a and B swapped. This is to ensure motors spin in same direction.

4 and 13 go to ROMI Driver BOARD LEFT  and 12 and 25 go to ROMI DRIVER BOARD RIGHT
so, to EXACTLY MATCH the SHIM. 4/13 and 25/12 need to be SWAPPED at U2 and U3.
and the AB swap in the header file is a feature and should not be made on the PCB.

madhephaestus, 2020-11-01:
The PCB should exactly match behavior of the 0.9.4 code running on the v3 shim. The same code running on a v3 shim must be interchangeable with the behavior of this module. The v3 shim was tested in terms of functionality and approved given the 0.9.4 software.

acamilo, 2020-11-01:
| 0.9.4 h file                          | shim |
| ----------------------------|-------|
| MOTOR_LEFT_PWM 13     |     13 |
| MOTOR_LEFT_DIR 4          |     4   |
|  MOTOR_RIGHT_PWM 12 |    12 |
| MOTOR_RIGHT_DIR 25    |     25  |
|  MOTOR_LEFT_ENCA 27  |     27  |
| MOTOR_LEFT_ENCB 26   |      26 |
| MOTOR_RIGHT_ENCA 32 |  14 ! |
| MOTOR_RIGHT_ENCB 14  | 32 ! |

Please confirm the swap on the right encoder is intentional.

madhephaestus, 2020-11-01:
the v2 and v3 are backwards, the v2 is top of the repo in 0.10.0, please do not look at 0.10.0 thats exists to deal with the errors this term. version 0.9.4 is correct.

acamilo, 2020-11-01:
Ok, I have checked 0.9.4 against the v3 shim.
and the v3 shim matches the edits in the v2 zen PCB.

I think we're all set with the motors and encoders.

closed 2020-11-01 by acamilo