WPIRoboticsEngineering/ZenDriver #29: IR receiver connected to an output only pin?

gcl8a, 2020-12-15:
The IR receiver is connected to pin 15. But the interface board page says that pin is output only.

Unless I'm mistaken about the pin functionality, this is a critical issue.

gcl8a, 2020-12-15:
After some testing, the pin works. Maybe this board has different functionality? I'll close this out, but I'm curious about the apparent contradiction.

madhephaestus, 2020-12-15:
Thats a good point actually, that pin is one of the strapping pins and needs to be held in 3.3v at boot time for the chip to boot.

gcl8a, 2020-12-15:
Luckily, the sensor is active low, so it’s generally high at upload.

Still, the reason I was looking into it was due to odd behavior. I did get it all working, but if there is a better solution, let’s do it.

madhephaestus, 2020-12-15:
For C term, it will just be on 15 and we tell them to plug the sensor in after programming, or that the sensor might interrupt programming if you send a signal while booting.

gcl8a, 2020-12-15:
And make it clear that if you’re having trouble uploading, one of the first things to try is removing the sensor.

acamilo, 2020-12-15:
The sensor is socketed, Instructing the students to remove it if they're having boot problems is what we'll have to do. In future revisions of the board we will switch it to GPIO35

madhephaestus, 2020-12-15:
Leave the issue open until the pin is swapped then